iic-jku/IIC-OSIC-TOOLS #161: Issue with the start_x.sh script on mac

claudiotalarico, 2025-08-17:
**Describe the bug**

After recently re-cloning the repository
`git clone --depth=1 https://github.com/iic-jku/iic-osic-tools.git`
I noticed that invoking the script
`./start_x.sh`
does not start the terminal as it should (or at least it used to)

Here is what I see:

```$ ./start_x.sh 
[INFO] Design directory auto-set to /Users/talarico/Google Drive/My Drive/eda/designs.
[INFO] Using indirect rendering.
[INFO] Container does not exist, creating iic-osic-tools_xserver_uid_986205870. Please be patient, this can take a few minutes ...

What's Next?
  1. Sign in to your Docker account → docker login
  2. View a summary of image vulnerabilities and recommendations → docker scout quickview hpretl/iic-osic-tools:latest
docker: invalid reference format: repository name must be lowercase.
See 'docker run --help'.
```

If I recycle the start_x.sh script from an old clone, the terminal starts and behave as expected.
The `start_x.sh` script I am using is [here](https://github.com/claudiotalarico/EE406/blob/main/start_x.sh)

I took a quick look at the two scripts differences, but could not figure out the reason for the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. clone the repository
`git clone --depth=1 https://github.com/iic-jku/iic-osic-tools.git`
2. `cd iic-osic-tools'
3. Edit the start_x.sh to set the DESIGN variable to the desired value
4. `./start_x.sh`

**Expected behavior**
Expecting the terminal to pop up

**Environment:**
 - OS: macOS
 - Operating mode: X11
 - Version tag: 2025.07

**Additional context**
The problem has been tested/replicated on both an Intel-based mac and on an Apple-Silicon based mac

claudiotalarico, 2025-08-19:
> @claudiotalarico<https://github.com/claudiotalarico> Thanks for testing this. One more thing: I faintly remember I also deinstalled and reinstalled Docker and Docker Desktop during our debug session. Would this be something you could try as well?

Hi Harald,
Thank you.
I tried to also de-installed and re-installed docker, (and Xquartz) but unfortunately still no success with the new script.

Note that the old script works with no problem (so from my perspective the issue is not a show stopper 😊)

I am attaching the script that I am using and that works in case you want to get to the bottom of the issue
(I am more than happy to do more testing if you have any more ideas to try).

Here are a few experiments/observations:
If I run the new script the image seem to get created just fine (although the terminal does not pop up).
At this point if I run the old script the terminal pops up and the DESIGNS directory gets mounted as it should.

If now I run the new script (and when asked I remove the x_server created by the old script), the terminal does not pop up
If I add this line (just above the OSTYPE checks) to the new script:

PARAMS="--security-opt seccomp=unconfined"

The terminal pops up (but the $DESIGNS directory is not mounted).
It seems like the definition of the PARAMS variable need some fixing w.r.t to what I did.

Thanks,
Claudio

MrHighVoltage, 2025-08-26:
Hi @claudiotalarico 

could you please run the failing and the working start script, additionally with `DRY_RUN=1`?

So like: `DRY_RUN=1 ./start_x.sh`. The commands should then be printed to the screen, but not actually executed. We can find out this way, where the issue originates from. Please copy the outputs here.

Thanks!

hpretl, 2025-08-27:
@claudiotalarico Maybe I have an idea: I think I also had a similar issue once when using drive locations with special characteris like space and so on. 

See: `/Users/talarico/Google Drive/My Drive/eda/designs`

Could you please try to use a location you mount without space?

closed 2025-08-27 by MrHighVoltage

claudiotalarico, 2025-08-27:
@MrHighVoltage @hpretl 
Thank you.
Just finished testing. It works like a charm :-)

claudiotalarico, 2025-11-23:
@MrHighVoltage and @hpretl 
The problem with design path with spaces seem to be back with 2025.11

Claudio

commit 2a704b7d (2025-08-27): Fix #161, a design path with space was not correctly quoted.