iic-jku/IIC-OSIC-TOOLS #165: Xschem copy and paste bug

slagernate, 2025-08-20:
It seems after initiating a copy (with `ctrl-c`) but then cancelling the copy with `esc`, a copy is pasted nonetheless on top of the original copied object(s).

**Screenshots**
https://github.com/user-attachments/assets/8634e317-f71e-4659-8e26-4aa8bc81c4da

**Environment:**
pulled hpretl/iic-osic-tools:latest today

**Additional context**
https://github.com/StefanSchippers/xschem/issues/387#issuecomment-3204495703

simi1505, 2025-08-24:
@slagernate, @hpretl and @StefanSchippers: I can confirm, something is not right here. The following happens with my Docker setup (07.2025):
- select a symbol
- press CTRL+C
- press CTRL+V (gray symbol appears, but do not place)
- press ESC (everything looks fine now, but...)
- select the original symbol
- press M for move and place somewhere else --> **another symbol appears!**

slagernate, 2025-08-25:
This does not occur with just text objects. Haven't tried other objects. Maybe it's something with how `esc` is handled in docker.

simi1505, 2025-08-25:
Strange, for me, the same behaviour happens with text objects also.

StefanSchippers, 2025-08-25:
I have now confirmed the bug. I was pressing esc before doing ctrl-V. Doing ctrl-V and then pressing esc leaves copied objects on top of previous.
WIll investigate and fix asap.

StefanSchippers, 2025-08-25:
The bug above can be avoided in older xschem releases by pressing Esc two times after a ctrl-V. First Esc aborts the move, second Esc cancels the paste operation.
Now both actions are done on a single Esc press.

hpretl, 2025-08-25:
@StefanSchippers I guess this is why I never experienced this issue. I am so conditioned to press ESC all the time, that I probably always hit ESC multiple times :-)