iic-jku/IIC-OSIC-TOOLS #186: librelane breaks at global placement step for design example dig/counter.v

ihagg, 2025-09-21:
**Describe the bug**
librelane breaks at global placement step for design example dig/counter.v:
_Classic - Stage 28 - Global Placement ... 27/79
ERROR: OpenROAD.GlobalPlacement failed with the following errors:
[RSZ-2001] failed bnet construction for _243_/Q
LibreLane will now quit_
In older tags, this example used to work nicely with openlane with the same json file.

**To Reproduce**
Steps to reproduce the behavior:
1. Copy examples/dig folder into designs folder
2. run "sak-pdk sky130A" 
3. run "librelane counter.json"
4. See error

**Expected behavior**
The complete flow was expected to end successfully and create the runs/final folder with the finalized design.

**Screenshots**
[image]

**Environment:**
 - OS: Linux
 - Operating mode: VNC
 - Version tag 2025.07

**Additional context**
librelane --smoke-test runs well, so it doesn't seem a librelane problem.

hpretl, 2025-09-21:
@ihagg This is (unfortunately) a known bug of LibreLane: https://github.com/librelane/librelane/issues/759

You can work around it by supplying a pin-order configuration.

ihagg, 2025-09-22:
@hpretl thank you for the hint, I was able to find a workaround with the pin order config file. I took me several try-error iterations to find the right syntax for the file. I am attaching you both the new "counter.json" and "pin_order.txt" in case you want to add them to the next image. It runs nicely with a simple "librelane counter.json".

[counter.json](https://github.com/user-attachments/files/22479125/counter.json)
[pin_order.txt](https://github.com/user-attachments/files/22479124/pin_order.txt)